sdr-enthusiasts/docker-piaware #2: arm64/v8

ChadOhman, 2019-06-27:
Any plans to build this for Pi 3B+?

`docker: no matching manifest for linux/arm64/v8 in the manifest list entries.`

```
ubuntu@dockerpi:~$ uname -a
Linux dockerpi 4.15.0-1038-raspi2 #40-Ubuntu SMP PREEMPT Thu Jun 6 07:33:02 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

mikenye, 2019-06-28:
I don't currently own a Pi 3B+, but I suppose I could purchase one and build for it.

Do you know if the new Pi 4B uses the same architecture as the Pi 3B+? If I'm going to buy something, I'd prefer to just buy the newer Pi 4B... :-)

mikenye, 2019-06-28:
It looks like the 4B also uses arm64/v8 (https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/README.md)

I'll buy one and build for that platform as soon as it arrives.

ChadOhman, 2019-06-28:
Hey Mike,
Yes, the 4B uses the same arm64/v8. Very excited to get my hands on one.
No rush, I was able to get piaware loaded up on obstruse's arm docker image. I appreciate it and I'll close the issue. Cheers!

mikenye, 2019-06-28:
OK, Pi4B ordered! Delivery expected 10th July.

mikenye, 2019-06-28:
I'm going to re-open and close once the arm64/v8 image is done (if that's ok).

ChadOhman, 2019-06-28:
Sounds great. Looking forward to it. Thanks Mike!

mikenye, 2019-08-06:
Just a quick update on this. I've (finally) received my RPi4B, however my MacBook power supply doesn't power it (so much for "standards" hey?) and the power supply I ordered is on backorder... :(

Sorry for the delay. Hopefully the PSU isn't far away...

mikenye, 2019-08-27:
Hi @ChadOhman, I finally received all the parts for my RPi4. I've created a dockerfile for aarch64, built the container on 64-bit Raspbian and updated the image. There should now be a manifest for linux/arm64/v8 that you can use. Please let me know how you fare.

closed 2019-08-27 by mikenye