sdr-enthusiasts/docker-piaware #15: No messages on ROCKPro64 (aarch64)

jondolan, 2020-04-14:
Hello! Thanks for your work on these great ADS-B images!

I have a working ADS-B setup (not in Docker) on a Pi 3B but I am looking to migrate my receiver to a ROCKPro64.

Everything appears to start up as expected (on both this image and the mikenye/readsb image), but there are 0.0 messages/sec in the PiAware GUI and likewise no decodings visible in the viewadsb tool.

I have tried all the USB ports (2xUSB2, 1xUSB3.0), to no avail. I also tried hooking the receiver up through a powered USB hub to eliminate potential power issues, no luck. Gain is left to the default.

Normally, even at the lowest times, I see ~50 messages/sec and as much as 200+. There are definitely planes in the area that I do properly decode if I move back to the Pi.

Please let me know how I can help!

Here is some debugging information:

lsusb:
```
...
Bus 003 Device 008: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
...
```

usb-devices, to show kernel module is properly blacklisted:
```
T:  Bus=03 Lev=02 Prnt=04 Port=01 Cnt=02 Dev#=  8 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=2838 Rev=01.00
S:  Manufacturer=Realtek
S:  Product=RTL2838UHIDIR
S:  SerialNumber=00000001
C:  #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
I:  If#= 1 Alt= 0 #EPs= 0 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
```

docker-compose : 
```
version: '2.0'

services:
  piaware:
    image: mikenye/piaware:latest
    tty: true
    container_name: piaware
    restart: always
    devices:
      - /dev/bus/usb/003/008:/dev/bus/usb/003/008
    ports:
      - 8080:8080
      - 30003:30003
      - 30005:30005
    environment:
      - TZ="?"
      - LAT=?
      - LONG=?
      - FEEDER_ID=?
```

startup output:
```
Creating piaware ... done
Attaching to piaware
piaware    | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
piaware    | [s6-init] ensuring user provided files have correct perms...exited 0.
piaware    | [fix-attrs.d] applying ownership & permissions fixes...
piaware    | [fix-attrs.d] done.
piaware    | [cont-init.d] executing container initialization scripts...
piaware    | [cont-init.d] 01-piaware: executing...
piaware    | Set feeder-id to ? in /etc/piaware.conf:1
piaware    | Set allow-mlat to yes in /etc/piaware.conf:2
piaware    | Set allow-modeac to yes in /etc/piaware.conf:3
[... 127 lines cut ...]
piaware    | [dump1090-fa] 0 surface position messages received                                                  
piaware    | [dump1090-fa] 0 airborne position messages received
piaware    | [dump1090-fa] 0 global CPR attempts with valid positions
piaware    | [dump1090-fa] 0 global CPR attempts with bad data
piaware    | [dump1090-fa]   0 global CPR attempts that failed the range check
piaware    | [dump1090-fa]   0 global CPR attempts that failed the speed check
piaware    | [dump1090-fa] 0 global CPR attempts with insufficient data
piaware    | [dump1090-fa] 0 local CPR attempts with valid positions
piaware    | [dump1090-fa]   0 aircraft-relative positions
piaware    | [dump1090-fa]   0 receiver-relative positions
piaware    | [dump1090-fa] 0 local CPR attempts that did not produce useful positions
piaware    | [dump1090-fa]   0 local CPR attempts that failed the range check
piaware    | [dump1090-fa]   0 local CPR attempts that failed the speed check
piaware    | [dump1090-fa] 0 CPR messages that look like transponder failures filtered
piaware    | [dump1090-fa] 0 non-ES altitude messages from ES-equipped aircraft ignored
piaware    | [dump1090-fa] 0 unique aircraft tracks
piaware    | [dump1090-fa] 0 aircraft tracks where only one message was seen
piaware    | [dump1090-fa] 0 aircraft tracks which were not marked reliable
piaware    | [dump1090-fa] CPU load: 10.7%
piaware    | [dump1090-fa]   211025 ms for demodulation
piaware    | [dump1090-fa]   144801 ms for reading from USB
piaware    | [dump1090-fa]   30267 ms for network input and background tasks
piaware    | [piaware] 0 msgs recv'd from dump1090 (0 in last 5m); 0 msgs sent to FlightAware
piaware    | [piaware] no new messages received in 3889 seconds, it might just be that there haven't been any aircraft nearby but I'm going to try to restart everything, just in case...
```

jondolan, 2020-04-14:
I have also built the Dockerfile manually, with the same results.

mikenye, 2020-04-14:
Hi @jondolan,

Can I get you to try the following:

Stop any containers/processes that might be using your RTL-SDR radio, then run the `readsb` container with the following command line:

```
docker run \
 -d \
 --rm \
 --name readsb \
 --privileged \
 -p 8080:80 \
 -p 30003:30003 \
 -p 30005:30005 \
 mikenye/readsb \
 --dcfilter \
 --device-type=rtlsdr \
 --fix \
 --forward-mlat \
 --json-location-accuracy=2 \
 --lat=YOUR_LATITUDE \
 --lon=YOUR_LONGITUDE \
 --mlat \
 --modeac \
 --ppm=0 \
 --net \
 --stats-every=3600 \
 --quiet
```

...swapping `YOUR_LATITUDE` and `YOUR_LONGITUDE` for the correct values.

Then, run the following command: `docker exec -it readsb viewadsb`.

I'd like to see if this gives you any output.

Thanks.

-Mike

jondolan, 2020-04-14:
Yes, Mike, I also tried your readsb image with no luck showing and decoded aircraft.

It is particularly weird because everything appears to be working fine, the dongle is detected properly, initialized, everything starts up properly, etc...but no decodings. I know the setup works, I just tested again on the Pi.

Also, I get a nginx 404 on port 8080 for the web interface in readsb.

This is what I got at startup of the readsb container:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Tue Apr 14 17:01:33 2020 UTC  Mictronics v3.8.2 starting up.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
Found Rafael Micro R820T tuner
rtlsdr: tuner gain set to 49.6 dB
Allocating 16 zero-copy buffers
```

jondolan, 2020-04-14:
I should also note I tried privileged mode and have run as root and a regular user.

Additionally, I tried copying the exact run parameters that are working on the Pi and manually start dump1090-fa with those, but no luck (modified the script in etc/services.d/piaware/run and manually rebuilt the timage).

`--device-index 0 --gain -10 --ppm 0 --max-range 360 --fix --net --net-heartbeat 60 --net-ro-size 1300 --net-ro-interval 0.2 --net-ri-port 0 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 1 --lat ? --lon ? --write-json /run/dump1090-fa --quiet)`

I also took off --quiet and I'm not seeing anything on the terminal.

This is probably a me-problem, so I will keep poking around. Was just wondering if you have seen anything like this before.

Thanks, ~ Jon

jondolan, 2020-04-14:
Compiling and running dump1090-fa outside of the container is working with
```
dump1090-fa --device-index 0 --gain -10 --ppm 0 --max-range 360 --fix --net --net-heartbeat 60 --net-ro-size 1300 --net-ro-interval 0.2 --net-ri-port 0 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 1 --lat ? --lon ? --write-json /run/dump1090-fa
```

and a stream of decodings is printed to the terminal.

mikenye, 2020-04-15:
Do you know what version of `rtlsdr` you're running? Both my `piaware` and `readsb` containers are using v0.6.0 - I wonder if it's an issue relating to that...

If you run the command `ldconfig -v | grep rtlsdr` it should output something like:

```
# ldconfig -v | grep rtl

        librtlsdr.so.0 -> librtlsdr.so.0.6.0
```

I'd be interested to see what version the working `dump1090-fa` system is using.

Cheers!

jondolan, 2020-04-16:
I have librtlsdr.so.0.5.3 installed.
I have it working outside the container on the ROCKPro64, so currently I have configed all the feeder images with an external feeder. I will try building rtlsdr v0.5.3 inside your container, my guess is that will work...

jondolan, 2020-04-16:
Yes, changing the Dockerfile to v0.5.3 tag worked for ROCKPro64 aarch64

the rtlsdr install section:

```
    echo "========== Install RTL-SDR ==========" && \
    git clone git://git.osmocom.org/rtl-sdr.git /src/rtl-sdr && \
    cd /src/rtl-sdr && \
    export BRANCH_RTLSDR=$(git tag --sort="-creatordate" | head -1) && \
    git checkout tags/v0.5.3 && \
    echo "rtl-sdr v0.5.3" >> /VERSIONS && \
    mkdir -p /src/rtl-sdr/build && \
    cd /src/rtl-sdr/build && \
    cmake ../ -DINSTALL_UDEV_RULES=ON -Wno-dev && \
    make -Wstringop-truncation && \
    make -Wstringop-truncation install && \
    cp -v /src/rtl-sdr/rtl-sdr.rules /etc/udev/rules.d/ && \
```

mikenye, 2020-04-17:
Ok, that's good news. It might be worth raising this issue with [the folks over at Osmocom](https://osmocom.org/projects/rtl-sdr/wiki) so they're able to fix the underlying issue with `rtl-sdr`.

In the meantime, I've thought about how to fix this issue with the container. I don't think I want to drop the version of `rtl-sdr` down, as this may introduce issues for other users.

I think for the time being, I'll have separately tagged docker images that have `rtl-sdr` version 0.5.3 for you and others that have the same hardware.

I'm currently building this now, stay tuned.

mikenye, 2020-04-17:
OK, you should be able to pull and use the docker image `mikenye/piaware:latest-rtlsdr-v0.5.3`.

I'll endevour to keep this updated for as long as I can, however ideally Osmocom should be made aware of the issue so they can fix and include in the next release of `rtl-sdr`.

Please let me know if `mikenye/piaware:latest-rtlsdr-v0.5.3` works for you.

Cheers.

mikenye, 2020-04-29:
Related: https://github.com/mikenye/docker-readsb/issues/3

mikenye, 2020-04-29:
Fixed in `latest` & `v3.8.1` image tags.

closed 2020-04-29 by mikenye