sdr-enthusiasts/docker-piaware #17: Docker image loads and checks in but continuously restarts dump1090

tmonfiletto, 2020-04-22:
LSUSB from host (raspiberrypi4 Ubuntu 19.10.01 arm64) Bus 001 Device 005: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
LSUSB from container: Bus 001 Device 005: ID 0bda:2838

[dump1090-fa] Wed Apr 22 10:15:16 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 6369 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
6370 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:15:17 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 6386 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
6387 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:15:18 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 6403 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
6404 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:15:19 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 6420 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
6421 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:15:20 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)

As you can see it restarts every second, i'm just not sure why

Here is the log from the start if it helps

[piaware] Connecting to FlightAware adept server at piaware.flightaware.com/1200
[piaware] Connection with adept server at piaware.flightaware.com/1200 established
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 286 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
287 Done | awk '{print "[dump1090-fa] " $0}'
[piaware] TLS handshake with adept server at piaware.flightaware.com/1200 completed
[piaware] FlightAware server certificate validated
[piaware] encrypted session established with FlightAware
[dump1090-fa] Wed Apr 22 10:25:16 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[piaware] adept reported location: 44.06796, -69.52292, 103ft AMSL
[piaware] logged in to FlightAware as user tmonfiletto
[piaware] my feeder ID is Redacted
[piaware] site statistics URL: https://flightaware.com/adsb/stats/user/tmonfiletto#stats-71046
[piaware] multilateration data requested
[piaware] no ADS-B data program is serving on port 30005, not starting multilateration client yet
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
[piaware] ADS-B data program 'dump1090' is listening on port 30005, so far so good
[piaware] Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30005 --stdout --lat 44.068 --lon -69.523
[piaware] Started faup1090 (pid 317) to connect to dump1090
[piaware] UAT support disabled by local configuration setting: uat-receiver-type
./run: line 48: 308 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
309 Done | awk '{print "[dump1090-fa] " $0}'
[piaware] lost connection to dump1090 via faup1090
[piaware] faup1090 exited normally
[piaware] will reconnect to dump1090 in 30 seconds
[dump1090-fa] Wed Apr 22 10:25:17 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 328 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
329 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:25:18 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
[dump1090-fa] Found Rafael Micro R820T tuner
[dump1090-fa] rtlsdr: enabling tuner AGC
[dump1090-fa] Allocating 4 zero-copy buffers
./run: line 48: 345 Killed exec "${DUMP1090_BIN}" "${DUMP1090_CMD[@]}" 2>&1
346 Done | awk '{print "[dump1090-fa] " $0}'
[dump1090-fa] Wed Apr 22 10:25:19 2020 EDT dump1090-fa starting up.
[dump1090-fa] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)

tmonfiletto, 2020-04-22:
Here is the output from my usb-devices, is it blacklisted properly?
T:  Bus=01 Lev=03 Prnt=03 Port=03 Cnt=02 Dev#=  5 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=2838 Rev=01.00
S:  Manufacturer=Realtek
S:  Product=RTL2838UHIDIR
S:  SerialNumber=00000001
C:  #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=usbfs
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)

tmonfiletto, 2020-04-22:
I pulled your other image

mikenye/piaware:latest-rtlsdr-v0.5.3

That seems better, its not just rebooting anymore

[piaware] mlat-client(279): Reconnecting in 30.0 seconds
[piaware] mlat-client(279): Beast-format results connection with 127.0.0.1:30104: connection established
[piaware] mlat-client(279): Input connected to localhost:30005
[piaware] mlat-client(279): Input format changed to BEAST, 12MHz clock
[piaware] 0 msgs recv'd from dump1090; 0 msgs sent to FlightAware
[piaware] ADS-B data program 'dump1090' is listening on port 30005, so far so good
[piaware] Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30005 --stdout --lat 44.068 --lon -69.523
[piaware] Started faup1090 (pid 302) to connect to dump1090
[piaware] piaware received a message from dump1090!
[piaware] piaware has successfully sent several msgs to FlightAware!

tmonfiletto, 2020-04-22:
You can mark this as resolved once you take a look at it, i'll pull any further information you need but as of now its working great with the updated image

mikenye, 2020-04-24:
Hi @tmonfiletto, sorry for the delay in responding to you - my day job has been quite busy.

This looks like it may be related to issue #15. There seems to be an incompatibility between `rtlsdr` version 0.6.0 and the RTL2838UHIDIR. For now, the fix is to use the image you found (tagged with `-rtlsdr-v0.5.3`).

I will find the time to open an issue with the maintainers of `rtlsdr` to see if they're able to fix this.

Thanks for getting in touch!

closed 2020-04-24 by mikenye