sdr-enthusiasts/docker-piaware #21: Dockerfile mlat reference incorrect

klara31, 2020-05-02:
Dockerfile, line 149 states:
```
ln -s /usr/bin/fa-mlat-client /usr/lib/piaware/helpers/ && \
```
however, this should be:
```
ln -s /usr/local/bin/fa-mlat-client /usr/lib/piaware/helpers/ && \
```
to get MLAT working

mikenye, 2020-05-02:
Ugh I'm an idiot. :-)
Thanks for picking this up.

mikenye, 2020-05-02:
Making the change has (obviously) fixed the problem:

```
# ps ax
<snip>
  314 ?        S      0:00 /usr/bin/python3 /usr/lib/piaware/helpers/fa-mlat-client --input-connect localhost:30005 --input-type dump1090 --results beast,connect,localhost:30104 --results beast,listen,30105 --results ext_basestation,listen,30106 --udp-transport x.x.x.x:xxxx:xxxxxxxxxx
<snip>
```

klara31, 2020-05-02:
Glad I could help (-:

While you're at it: when I enter the container and issue the command piaware-status, I guess it is missing pgrep. Do you want a new issue for that?

mikenye, 2020-05-02:
Nope, that's ok, I'll fix that under this issue too. :-)

mikenye, 2020-05-02:
Interestingly it looks like the `fa-mlat-client` issue is related to moving from Alpine to Debian, which happened on the 29th April 2020. Looking at my FlightAware My ADSB, I can see my MLAT dropped to 0 from 30th April onwards.

mikenye, 2020-05-02:
Blame: ebec3721f0ad69add4d052608a4a86279dc8d8d5

klara31, 2020-05-02:
Just wondering: why are you using the MLAT client from mutability and not the one from FlightAware?

mikenye, 2020-05-02:
FlightAware appear to use mutability/mlat-client.

See: https://github.com/flightaware/piaware_builder/blob/66a92f1e4b267bb8113ff7ff0e3415b6c7f8a33d/sensible-build.sh#L89

mikenye, 2020-05-02:
OK, adding the `procps` package seems to have fixed `piaware-status`:

```
$ docker exec -it piaware piaware-status
PiAware master process (piaware) is running with pid 257.
PiAware ADS-B client (faup1090) is running with pid 304.
PiAware ADS-B UAT client (faup978) is not running (disabled by configuration settings)
PiAware mlat client (fa-mlat-client) is running with pid 301.
Local ADS-B receiver (dump1090) is running with pid 276.

socat (pid 250) is listening for ES connections on port 30005.
faup1090 is NOT connected to the ADS-B receiver.
piaware is connected to FlightAware.

dump1090 is producing data on localhost:30005.

Your feeder ID is xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx (configured at /etc/piaware.conf:1)
```

klara31, 2020-05-02:
Great! A new build will be pushed to docker automatically?

mikenye, 2020-05-02:
I'm building it right now. amd64 & arm/v7 are done, just waiting on arm64...

klara31, 2020-05-02:
nice

mikenye, 2020-05-02:
OK! `mikenye/piaware:latest` and `mikenye/piaware:v3.8.1` have been built and pushed. Let me know how you go.

klara31, 2020-05-02:
It's working!

Just switched from bare metal to docker on my Pi, I will let you know if I bump into other issues.

mikenye, 2020-05-02:
Yay. Thanks, please do. If you have Discord feel free to message me on there. mikenye#4103.

closed 2020-05-02 by mikenye