sdr-enthusiasts/docker-piaware #24: Suggestion: redirect directly to FlightAware user statistics page

klara31, 2020-05-05:
When clicking on the FlightAware logo on the top right, you are redirected to the FlightAware home page. It's a lot easier when you are directly redirected to your personal statistics page. When running locally, I was using the following patch:
```
--- /usr/share/dump1090-fa/html/index.html	2019-08-01 10:33:43.000000000 +0200
+++ /usr/share/dump1090-fa/html/index.html	2019-08-13 22:47:39.299981896 +0200
@@ -48,7 +48,7 @@
 
 		<div id="header" class="infoHeading">
 			<div class="logoContainer">
-				<a href="http://flightaware.com/" target="_blank"><img src="images/fa_logo_color.png" alt="FlightAware" class="flightawareLogo" srcset="images/fa_logo_color.png 1x, images/fa_logo_color@2x.png 2x, images/fa_logo_color@3x.png 3x"></a>
+				<a href="https://flightaware.com/adsb/stats/user/xxxxxxxxxxxxx" target="_blank"><img src="images/fa_logo_color.png" alt="FlightAware" class="flightawareLogo" srcset="images/fa_logo_color.png 1x, images/fa_logo_color@2x.png 2x, images/fa_logo_color@3x.png 3x"></a>
 				<img src="images/pa-sa-logo.png" alt="PiAware SkyAware" class="adsbLogo piAwareLogo" srcset="images/pa-sa-logo.png 1x, images/pa-sa-logo@2x.png 2x, images/pa-sa-logo@3x.png 3x">
 				<img src="images/ff-sa-logo.png" alt="FlightFeeder SkyAware" class="adsbLogo flightfeederLogo" srcset="images/ff-sa-logo.png 1x, images/ff-sa-logo@2x.png 2x, images/ff-sa-logo@3x.png 3x" style="display: none;">
 			</div>
```
Is it possible to implement this, probably using an environment variable on this image?

klara31, 2020-05-05:
Awesome! Thanks :)

mikenye, 2020-05-06:
No worries! Did it work for you?

Also, I've been googling about using a Bing API key with `piaware`... Do you have any info on this or suggestions on how to implement?

klara31, 2020-05-06:
config.js has a line for that:
```BingMapsAPIKey = "xxxxxxxxxxx";```
I already tried adding this line to config.js and that works OK.

mikenye, 2020-05-06:
OK, the `latest` tag of the image should allow you to set `BINGMAPSAPIKEY`. Let me know how you fare.

klara31, 2020-05-06:
Yep, working! Thanks again!

mikenye, 2020-05-06:
You're most welcome!

klara31, 2020-07-22:
The redirect to the user statistics page is gone...

mikenye, 2020-07-22:
I'll check it out

mikenye, 2020-07-22:
Ok this should now be fixed. Apologies for the inconvenience.

klara31, 2020-07-22:
Confirmed! Thanks (-:

closed 2020-07-22 by mikenye