sdr-enthusiasts/docker-piaware #31: Exec format error when running container on Raspberry Pi 3B

seanauff, 2020-06-08:
I'm getting the exec format error in the logs when attempting to run the `latest` tag on my Raspberry Pi 3B.
```
$ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

`latest` tag runs fine on my Atomic Pi (x86_64), so there seems to be something amiss with the way the arch gets detected. 

I was able to build the image myself on the Pi 3B and it works great.

echoz, 2020-06-08:
Seeing this for the following images too:

- [planefinder](https://github.com/mikenye/docker-planefinder)
- [flightradar24](https://github.com/mikenye/docker-flightradar24)
- [radarbox](https://github.com/mikenye/docker-radarbox)

mikenye, 2020-06-09:
Acknowledging this. Please accept my apologies - something has gone awry in my build farm. I'm actively troubleshooting this and hope to have it resolved shortly.

echoz, 2020-06-09:
No worries. Thanks for maintaining this! Just thought you'd like to know!

mikenye, 2020-06-09:
Appreciate the heads up! :-)

mikenye, 2020-06-09:
@echoz @seanauff all images should be fixed now!

seanauff, 2020-06-09:
Tested and working, thanks!

closed 2020-06-09 by seanauff