sdr-enthusiasts/docker-piaware #34: Log messages do not contain a timestamp

jheinitz, 2020-06-19:
Hello,

I like your image and it makes it very easy to setup a feeder for FlightAware. I'm using a docker-compose setup which makes it even more easier to maintain for me. However, when I look at the output from docker-compose logs, I do not see any timestamps. Is it possible to change that somewhere or is it an issue of the piaware software itself? Please see my log for reference:

`user1@server1:/opt/docker-piaware$ docker-compose logs --tail=5`
`Attaching to piaware`
`piaware    | [piaware] 268 msgs recv'd from dump1090 (78 in last 5m); 268 msgs sent to FlightAware`
`piaware    | [piaware] 292 msgs recv'd from dump1090 (24 in last 5m); 292 msgs sent to FlightAware`
`piaware    | [piaware] 319 msgs recv'd from dump1090 (27 in last 5m); 319 msgs sent to FlightAware`
`piaware    | [piaware] 324 msgs recv'd from dump1090 (5 in last 5m); 324 msgs sent to FlightAware`
`piaware    | [piaware] 407 msgs recv'd from dump1090 (83 in last 5m); 407 msgs sent to FlightAware`
`user1@server1:/opt/docker-piaware$`

Best regards

Jens

mikenye, 2020-06-22:
Hi Jens,

Unfortunately, `piaware` doesn't include the date/time in their log messages. However, I pipe the logs through `awk` to add the `[piaware]` text, so I've modified the `awk` syntax to include the current date/time.

If you re-pull the image, you should see the date/time included for the `piaware` logs.

Let me know if this solves your problem.

Kind regards,
Mike Nye

jheinitz, 2020-06-22:
Hi Mike,

well done. That's looks good now. The only drawback is that messages are
not shown immediately, but after a long period of time. It seems that the
logging is now buffered. Could you add something to remove the buffering of
the logs?

Thanks and best regards

Jens

mikenye, 2020-06-22:
OK, I've changed the buffering for the `awk` instances. Can you re-pull and try again?

jheinitz, 2020-06-22:
Hi Mike,

that's perfect. Logs appear immediately and timestamp is included.
Thanks a lot for the quick fix.

Cheers

Jens

mikenye, 2020-06-22:
No problems, thanks!

closed 2020-06-22 by mikenye