sdr-enthusiasts/docker-piaware #49: Connecting YAAC Client to piaware ADSB output

n8acl, 2020-07-24:
Hello!

This is less an issue, more of a question. I am a Ham Radio Operator and I am trying to tie the Yet Another APRS Client (YAAC) program to the ADS-B output from the PiAware Container. YAAC has a plugin that will allow the software to decode ADS-B from dump1090 and display it on a map with other APRS Stations. Trying to do this for a couple of reasons: 1) it's geeky and cool and 2) trying to tidy up a screen I use to display dashboards and maps and stuff. If I can combine two things into the one app that would be awesome.

I have been able to get YAAC to connect to port 30002 but it does not display the planes on the map, even when they are showing on the skyaware map. 

So I looked on the YAAC website and I found this page: https://www.ka2ddo.org/ka2ddo/YAACdocs/config_adsb.html on configuring ADS-B. It says it needs the ASCII-hex frames to decode properly.

So my question: Does the PiAware docker container dump the ASCII-hex data to a port (and which port) or is there a way to enable it or something else I am missing? If you need anything from me, please let me know. It's very possible it's a terminology thing and I am sitting here staring at it and just not computing....

Thanks for all your hard work with the container. I am running it on a RPi4 2gig and it runs great. Have it running for a while now and feeding to Flightaware. Just added the FR24feed container tonight. 

Jeff

n8acl, 2020-07-24:
HA never mind. I was able to get it to work with your readsb container. Thanks!

closed 2020-07-24 by n8acl

mikenye, 2020-07-29:
Hi @n8acl, no problems! I'm glad you were able to find a solution.