sdr-enthusiasts/docker-piaware #60: Healthcheck now failing.  Missing healthcheck script.

scottwallacesh, 2020-11-07:
```
# docker inspect piaware | jq .[].State.Health.Log[-1].Output 
"/bin/sh: 1: /healthcheck.sh: not found\n"
```

mikenye, 2020-11-07:
Hi @scottwallacesh,

The current version of the container has the healthcheck commented out (while I work on it): <https://github.com/mikenye/docker-piaware/blob/dfa8dbda3f43f2338ea769599ececd862d94f994/Dockerfile#L281>

Are you manually specifying a healthcheck?

If so, I'd recommend disabling it for a few days while I re-write the healthcheck.

Sorry for the inconvenience.

-Mike

mikenye, 2020-11-07:
FYI - I'd recently made a significant amount of changes in order to properly implement UAT (dump978/skyaware978), which broke the healthcheck. Rather than delay the release of the image by rewriting the healthcheck code, I figured I'd release the new version with the healthcheck disabled, and finish the healthcheck over the next day or two.

scottwallacesh, 2020-11-07:
> Are you manually specifying a healthcheck?

No.  I'm simply using `mikenye/piaware` from Docker Hub:
```
# docker image pull mikenye/piaware:latest 
latest: Pulling from mikenye/piaware
Digest: sha256:dc21f1cf2aab5758b5940a9d0eba20c8b75fd1483891f0cfa475972f2329054b
Status: Image is up to date for mikenye/piaware:latest
docker.io/mikenye/piaware:latest
```

mikenye, 2020-11-07:
Can you run `uname -m` on the host with the problem so I can test?

scottwallacesh, 2020-11-07:
Ahh... it appears that `docker-compose` hangs on the the concept of a healthcheck until the container is removed and recreated.  Simply recreating with a new image wasn't enough as it still said that a healthcheck existed.

Sorted now.

scottwallacesh, 2020-11-07:
> Can you run `uname -m` on the host with the problem and post the output so I know which architecture to test on?

```
# uname -m
x86_64
```

closed 2020-11-07 by scottwallacesh

mikenye, 2020-11-07:
Oh that's really annoying. I didn't realise.

I'll try and get the healthcheck finished ASAP.