sdr-enthusiasts/docker-piaware #67: Docker-piaware not recieving messages from readsb-protobuf

hjboven, 2020-12-23:
Hi Mike

Just discovered your images yesterday and the seem to build an extremly nice environment to replace my standalone Piaware feeder.  I'm using a Raspberry Pi 3b+ with Ubuntu Server 20.04 for Rasberry on it. Works like a charm.

So far I've got readsb-protobuf installed and it seem to work fine. 
"docker exec -it readsb viewadsb" shows me a nice Table with plenty of recieved messages. Also the graphical output on port 8080 works just fine.

But when I start docker-piaware it reports
"0 msgs recv'd from dump1090 (0 in last 5m); 0 msgs sent to FlightAware"

Somehow I do not seem to get the connection working between the Containers.
I'm using BEASTHOST | 127.0.0.1 but when i change it to the ip address of the Host the result stays the same.

Do you happen to know what I am missing from your manual in Github??

regards,
Henk

mikenye, 2020-12-23:
Hi Henk,

Are you using docker-compose? If so, you should specify the container name instead of a loopback address for the `BEASTHOST` variable.

Without docker, if you were running these binaries on the host you’d be absolutely correct. However, when running inside a container, the loopback address points to the local container and not the underlying host.

If you called your readsb container “`readsb`”, and you’re using docker-compose as-per the guide, set `BEASTHOST=readsb`. Docker will provide name resolution so that `readsb` always resolves to the `readsb` container. This works because when using docker-compose, it places all your containers into their own docker network. Docker will perform name resolution for containers when on the same network, so you can then set `BEASTHOST=readsb`.

If you aren’t using docker-compose, you have a couple of options:
- Create a docker network (eg `docker network create adsbnet`) and add `--network=adsbnet` to your `docker run` command. Docker will perform name resolution for containers when on the same network, so you can then set `BEASTHOST=readsb`.
- Map the BEAST and SBS ports through to the `readsb` container (eg `-p 30003:30003 -p 30005:30005`), then set `BEASTHOST` to the LAN IP of the docker host.

I hope this helps!

Kind regards,
Mike

hjboven, 2020-12-29:
Hello Mike,

I was not using Docker-Compose to be honest have not used Docker-Compose before. I set it up now using Docker-Compose folowing your Gitbook and it all works. Thank you very much

closed 2020-12-29 by hjboven

mikenye, 2020-12-30:
Hi @hjboven, I'm glad you got it working mate!