sdr-enthusiasts/docker-piaware #71: port and IP as shown on flightaware site information

mo8Zomo0, 2021-03-13:
Hi Mike&all, 

first of all thx for the great work and documentation. I found some minor thing to improve:

https://flightaware.com/adsb/stats/user/<user> shows the internal IP and port of the local webinterface
under "site information": "Web Interface: View live data (requires local network connection)"
which shows as: http://172.17.0.6:8080/

This is the internal docker IP address or piaware:
# docker inspect -f '{{range.NetworkSettings.Networks}}{{.IPAddress}}{{end}}' piaware
172.20.0.6

As I assume that at least the IP is reported from the piaware image somehow, it might be nice if this could get an option, 
like default(current behaviour), auto-detect (which could show the main locally routed ip and the correct port (here 8081), or manual, where you can change it to what you want.

mikenye, 2021-03-15:
Hi @mo8Zomo0,

I'm not sure I can implement this without making changes to the piaware source code.

Piaware gets the host's IP address by getting the output of the command `ip -o address show dev $dev`.

It appears to be this section in the piaware source code:
https://github.com/flightaware/piaware/blob/515841346eb6783d1bf9553675aef988fe64748b/package/fa_sysinfo.tcl#L197

There doesn't appear to be a setting that overrides this.

Did you have any thoughts on how we could achieve this outside of modifying the piaware source code?

mo8Zomo0, 2021-03-15:
@mikenye no, not aware how to archive this without patching piaware.
So maybe (as its not really a mikenye issue, but a general docker/piaware issue, I opened a case
towards: https://github.com/flightaware/piaware/issues/70
So lets see.

thx!

mikenye, 2021-03-22:
PR opened: https://github.com/flightaware/piaware/pull/71