sdr-enthusiasts/docker-piaware #73: Update web interface to Skyaware

mikenye, 2021-03-15:


klara31, 2021-03-16:
Just updated the container, still getting a ```404 Not Found``` when surfing to /skyaware ...

mo8Zomo0, 2021-03-16:
I also just pulled to the latest and had my :8081 website broken. 

It looks to me the path is wrong. There is only a folder /usr/share/skyaware978
but /etc/lighttpd/skyaware.lighttpd.conf points to /usr/share/skyaware as well:

>   "/dump1090-fa/" => "/usr/share/dump1090-fa/html/",
>   "/skyaware/" => "/usr/share/skyaware/html/"
>     "/" => "/usr/share/skyaware/html/"
>   "/skyaware978/" => "/usr/share/skyaware978/html/"
>     "/" => "/usr/share/skyaware978/html/"

a hotfix to get it going might be:
/opt/adsb# _docker exec -it piaware ln -s /usr/share/skyaware978 /usr/share/skyaware_    
worked at least for me.

Which path is expected to be there the one with the number or the one without?

klara31, 2021-03-22:
It's working OK now

mikenye, 2021-03-22:
Thanks @klara31. Sorry for the inconvenience!

closed 2021-03-22 by mikenye