sdr-enthusiasts/docker-piaware #77: buffer overflow detected v5

visibilityspots, 2021-03-20:
when using the images both v4 / v5  I got a buffer overflow detected and a core dump running on a pi 4 did I overlooked something?

```
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [dump1090] 2021/03/20 16:18:20 Net-only mode, no SDR device or file open.
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [services.d] done.
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [piaware] 2021/03/20 16:18:20 ****************************************************
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [piaware] 2021/03/20 16:18:20 piaware version 5.0 is running, process ID 339
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [piaware] 2021/03/20 16:18:20 your system info is: Linux 954d8927f124 5.10.14-2-ARCH #1 SMP Fri Feb 12 13:51:38 UTC 2021 armv7l GNU/Linux
Mar 20 16:18:20 node08 FLIGHTAWARE[317]: [piaware] 2021/03/20 16:18:20 *** buffer overflow detected ***: /usr/bin/piaware terminated
Mar 20 16:18:21 node08 FLIGHTAWARE[317]: ./run: line 19:   339 Aborted                 (core dumped) "${PIAWARE_BIN}" "${PIAWARE_CMD[@]}" 2>&1
Mar 20 16:18:21 node08 FLIGHTAWARE[317]:        341 Done                    | stdbuf -o0 awk '{print "[piaware] " strftime("%Y/%m/%d %H:%M:%S", systime()) " " $0}'
```

mikenye, 2021-03-22:
Hi Jan,

Looking into this, I can see that if IPv6 is disabled by inserting `ipv6.disable=1` into `/boot/cmdline.txt`, then piaware will undergo a buffer overflow:

- https://discussions.flightaware.com/t/piaware-3-6-2-buffer-overflow-detected/40089/8
- https://github.com/flightaware/piaware/issues/51
- https://www.bountysource.com/issues/70265648-buffer-overflow-when-ipv6-gets-disabled

Have you disabled IPv6?

visibilityspots, 2021-03-22:
ah damn, yes on that node I have it disabled in the past for some tests. :facepalm: 

thanks!

closed 2021-03-22 by visibilityspots

mikenye, 2021-03-22:
Glad it was an easy fix! :-)