sdr-enthusiasts/docker-piaware #82: Does the __arm__ flag get defined in BuildX?

luke-iqt, 2021-06-16:
I am trying to get a sense of how much latency there is on the receive side using Dump1090 on a Pi4. I noticed this comment in the RTL-SDR code:
https://github.com/flightaware/dump1090/blob/f862b530c8cea51f4bb9e72356579be67818dcaf/sdr_rtlsdr.c#L56-L59

I was wondering if the `__arm__` flag gets defined in the BuildX system from arm/v7 images? 

The commit message explains a bit more: 
https://github.com/flightaware/dump1090/commit/653ad6127a57b88af78b30e9790bb97f80f8eff7

I am trying to use ADS-B messages to take picture of aircraft as they fly by, so I am trying to account for all the latency in the messages: https://github.com/IQTLabs/SkyScan

mikenye, 2021-07-04:
Hi @luke-iqt,

I don't know the answer to your question, however when the build process runs it is run using QEMU emulating the target hardware. So for all intents and purposes, it should be an ARM system.

I added a `uname -a` into the Dockerfile of a temporary branch, and created a draft PR to allow github actions to run. Results for the test builds for each arch are as follows:

- i386: https://github.com/mikenye/docker-piaware/pull/83/checks?check_run_id=2983484597#step:6:3275
- amd64: https://github.com/mikenye/docker-piaware/pull/83/checks?check_run_id=2983484556#step:6:3267
- arm64: https://github.com/mikenye/docker-piaware/pull/83/checks?check_run_id=2983484570#step:6:3255
- arm/v7: https://github.com/mikenye/docker-piaware/pull/83/checks?check_run_id=2983484585#step:6:3612
- arm/v6: https://github.com/mikenye/docker-piaware/pull/83/checks?check_run_id=2983484577#step:6:3623

If there's any way I can help you answer this question please let me know. I am not a C programmer so I'm not sure how else to answer your question without further gudance.

mikenye, 2021-08-24:
Closing due to inactivity

closed 2021-08-24 by mikenye