sdr-enthusiasts/docker-piaware #87: PiAware v6.0 adaptive gain

emilburzo, 2021-09-04:
(The release announcement email barely came out and you already updated the docker images, amazing!)

One of the highlights of the v6 release seems to be the [adaptive gain](https://github.com/flightaware/dump1090/blob/master/README.adaptive-gain.md).

What would be the easiest way to try it out with the docker image?

mikenye, 2022-03-28:
Hi @emilburzo, sorry for the delay in implementing this.

PR #124 implements the following:

> The following settings control the [adaptive gain configuration](https://github.com/flightaware/dump1090/blob/master/README.adaptive-gain.md) of `dump1090` when using `RECEIVER_TYPE=rtlsdr`.
> 
> | Environment Variable | Possible Values | Description | Default |
> | -------------------- | --------------- | ------- | ------- |
> | `DUMP1090_ADAPTIVE_RANGE` | `true` or unset | Set to any value to adjust gain for target dynamic range. | |
> | `DUMP1090_ADAPTIVE_RANGE_TARGET` | A value in dB | Set target dynamic range in dB. | |
> | `DUMP1090_ADAPTIVE_BURST` | `true` or unset | Set to any value to adjust gain for too-loud message bursts. | |
> | `DUMP1090_ADAPTIVE_MIN_GAIN` | A value in dB | Set gain adjustment range lower limit (dB). | |
> | `DUMP1090_ADAPTIVE_MAX_GAIN` | A value in dB | Set gain adjustment range upper limit (dB). | |
> | `DUMP1090_ADAPTIVE_DUTY_CYCLE` | A percentage | Set adaptive gain duty cycle % (1..100) </br> See [Reducing the CPU cost of adaptive gain](https://github.com/flightaware/dump1090/blob/master/README.adaptive-gain.md#reducing-the-cpu-cost-of-adaptive-gain) | |

mikenye, 2022-03-28:
Once PR #124 is merged, this issue will automatically be closed. After this, the new image will build (will take about 1-2 hours). Once complete, please pull the image and update your container.

Please feel free to re-open the issue if you're still having problems after that.

Thanks.

PR #124 (2022-03-28): Fixes for 7.2
- Add symlink for skyaware978 to fix 404 error. Fixes #112 
- Start dump1090 if `RECEIVER_TYPE` set (but not `DUMP1090_DEVICE`). Fixes #123
- Add `MLAT_RESULTS_BEASTHOST` and `MLAT_RESULTS_BEASTPORT`. Fixes #89 
- Add environment variables to control `dump1090` adaptive gain. Fixes #87 
  - `DUMP1090_ADAPTIVE_RANGE`, `DUMP1090_ADAPTIVE_RANGE_TARGET`, `DUMP1090_ADAPTIVE_BURST`, `DUMP1090_ADAPTIVE_MIN_GAIN`, `DUMP1090_ADAPTIVE_MAX_GAIN`, `DUMP1090_ADAPTIVE_DUTY_CYCLE`
closed 2022-03-28 by mikenye