sdr-enthusiasts/docker-piaware #89: MLAT in relay mode

ZeeBOB, 2021-09-12:
I'm running dump1090-fa on a remote machine. When I have this container in relay mode MLAT is unable to connect, probably because it's trying to connect to dump1090-fa on localhost:30104, as per the Multilateration Settings on [FlightAware's Advanced Configuation](https://flightaware.com/adsb/piaware/advanced_configuration) page:
> The default settings set up three items:
> - beast,connect,localhost:30104 - this feeds multilateration results back to the local dump1090 for display on the map

If the container is in relay mode, it should reuse the BEASTHOST variable and add the following line to the piaware.config file:
`mlat-results-format beast,connect,$BEASTHOST:30104`
I'm guessing this is best done in the **docker-piaware/rootfs/etc/cont-init.d/01-piaware** file
`piaware-config mlat-results-format "beast,connect,${BEASTHOST}:30104"`

mikenye, 2022-03-28:
Hi @ZeeBOB, sorry for the delay in getting back to you about this.

When using the container in relay mode (with `BEASTHOST` set), the container behaves as follows:

- `dump1090` is started, and listens on `0.0.0.0:30004` for incoming BEAST connections.
- `beast-splitter` is started, and configured to connect to `$BEASTHOST:$BEASTPORT`, receives BEAST data, connects to `dump1090` (via `127.0.0.1:30004`) and sends BEAST data.
- `piaware` starts `mlat-client`, which is configured to receive data from `dump1090` (via `127.0.0.1:30005`).
- `mlat-client` serves data on `0.0.0.0:30105`.

As `dump1090` doesn't appear to have a `--net-connector` (like `readsb` does), I see your predicament... Users of `readsb` can use a `--net-connector` to pull in MLAT results, however it looks like there's no functionality in `dump1090` to do this. 

Would you accept me adding an `MLAT_RESULTS_BEAST_CONNECT` or similar environment variable allowing the `mlat-results-format` piaware config to be set manually?

mikenye, 2022-03-28:
Hi @ZeeBOB,

I've added:

| Environment Variable | Possible Values | Description | Default |
| -------------------- | --------------- | ------- | ------- |
| `MLAT_RESULTS_BEASTHOST` | a hostname or IP | Specify an external host where MLAT results should be sent. | |
| `MLAT_RESULTS_BEASTPORT` | a port number | Specify the TCP port number where MLAT results should be sent. | `30104` |

These control the `mlat-results-format` setting:

```bash
# If MLAT_RESULTS_BEAST_CONNECT is specified
if [[ -n "$MLAT_RESULTS_BEASTHOST" ]]; then
  piaware-config mlat-results-format "$(piaware-config -show mlat-results-format) beast,connect,${MLAT_RESULTS_BEASTHOST}:${MLAT_RESULTS_BEASTHOST:-30104}"
fi
```

Will be merged in #124.

mikenye, 2022-03-28:
Once PR #124 is merged, this issue will automatically be closed. After this, the new image will build (will take about 1-2 hours). Once complete, please pull the image and update your container.

Please feel free to re-open the issue if you're still having problems after that.

Thanks.

PR #124 (2022-03-28): Fixes for 7.2
- Add symlink for skyaware978 to fix 404 error. Fixes #112 
- Start dump1090 if `RECEIVER_TYPE` set (but not `DUMP1090_DEVICE`). Fixes #123
- Add `MLAT_RESULTS_BEASTHOST` and `MLAT_RESULTS_BEASTPORT`. Fixes #89 
- Add environment variables to control `dump1090` adaptive gain. Fixes #87 
  - `DUMP1090_ADAPTIVE_RANGE`, `DUMP1090_ADAPTIVE_RANGE_TARGET`, `DUMP1090_ADAPTIVE_BURST`, `DUMP1090_ADAPTIVE_MIN_GAIN`, `DUMP1090_ADAPTIVE_MAX_GAIN`, `DUMP1090_ADAPTIVE_DUTY_CYCLE`
closed 2022-03-28 by mikenye