sdr-enthusiasts/docker-piaware #102: realtime clock inaccessible on v7.1 image

dervogt, 2022-01-20:
Hi Mike,

it seems there is an issue with the recent v7.1 image and access tot he system time after upgrading my piaware image to the "latest" tag.

The piaware container throws a lot of these errors and CPU load is incredibly high:
sleep: cannot read realtime clock: Operation not permitted

WHen reverting to the docker hub tag / image mikenye/piaware:v6.1 the error message goes away and CPU load is back to normal again.

I am running the container in a docker stack, these are my settings from the stack compose file:

_piaware:
    image: mikenye/piaware:latest
    tty: true
    container_name: PiAware
    hostname: piaware
    restart: always
    environment:
      - TZ=Europe/Berlin
      - LAT=53.68030
      - LONG=10.03456
      - FEEDER_ID=REMOVED FOR GITHUB
      - RECEIVER_TYPE=rtlsdr
      - BEASTHOST=adsb_rcvr_ham
      - BEASTPORT=30005
      - VERBOSE_LOGGING=false
      - ALLOW_MODEAC=yes
    networks:
      - ADSB
    healthcheck:
      disable: true
    tmpfs:
      - /run:exec,size=64M
      - /var/log_
      - 
Let me know if you need anything else. 

Running the container in privileged mode seems to be a possible solution as per the web, but this is not supported when running in Stack as far as I know.

fredclausen, 2022-01-20:
Mike, I'm not sure your intentions but the image probably needs to be reverted back to `buster-20211220-slim` to fix the RTC issue (technically libseccomp2 on the host) or `bullseye-20211220-slim` and provide instructions to use the script that @kx1t made (https://github.com/fredclausen/docker-acarshub/blob/main/arm32.MD / https://raw.githubusercontent.com/fredclausen/docker-acarshub/main/libseccomp2-checker.sh) to fix the issue on the user's host.

mikenye, 2022-01-21:
Hi @fredclausen, thanks so much for this. I will update this image's README.md to include the wording from your repo and comment above. :-)

fredclausen, 2022-01-21:
You're welcome Mike! If it helps you I can move the ARM32 libseccomp2 install stuff in to a new repo and make a more generic readme explaining the issue and the fix.

mikenye, 2022-01-21:
@dervogt, further to the discussions above, the TLDR is here: https://github.com/fredclausen/Buster-Docker-Fixes. That repo explains everything and has a workaround/fix.

closed 2022-01-21 by mikenye