sdr-enthusiasts/docker-piaware #127: Incorrect Feeder ID Being Used

jasonkguinn, 2022-03-29:
As of around 8:15am EDT, my PiAware container stopped supplying data to my correct FlightAware feeder ID. I do have watchtower performing auto-updates to the containers. No config changes have been made recently and all other feeders are functioning nominally. Environment variables and logs from piaware are below. Please let me know if I can provide any other information that might be helpful - and thank you in advance!

```
FEEDER_ALT_FT=1730
FEEDER_ALT_M=527
FEEDER_LAT=36.339410
FEEDER_LONG=-82.383130
FEEDER_TZ=America/New_York
ADSBX_UUID=5dd808dc-849f-11ec-93e8-0242ac110005
ADSBX_SITENAME=jguinn-johnson_city_1
PIAWARE_FEEDER_ID=16b0b6bc-01d7-454d-8319-2e41976bc832
FR24_SHARING_KEY=a750b015a2f6ac0f
FR24_RADAR_ID=T-KTRI17
PLANEFINDER_SHARECODE=61fec0122f5e7
RADARBOX_SHARING_KEY=7fb14cd3417106dd3f23831d73a4364b
DUMP1090_DEVICE=1090
```

```
piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[353] N socket 1 (fd 6) is at EOF
piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[353] N exiting with status 0
piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[330] N childdied(): handling signal 17
piaware       | [beast-splitter] 2022/03/28 19:11:37 net(readsb:30005): i/o error: End of file
piaware       | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
piaware       | [s6-init] ensuring user provided files have correct perms...exited 0.
piaware       | [fix-attrs.d] applying ownership & permissions fixes...
piaware       | [fix-attrs.d] done.
piaware       | [cont-init.d] executing container initialization scripts...
piaware       | [cont-init.d] 00-libsecomp2: executing... 
piaware       | [cont-init.d] 00-libsecomp2: exited 0.piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[353] N socket 1 (fd 6) is at EOF
piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[353] N exiting with status 0
piaware       | [978json-splitter] 2022/03/28 19:11:36 2022/03/28 19:11:36 socat[330] N childdied(): handling signal 17
piaware       | [beast-splitter] 2022/03/28 19:11:37 net(readsb:30005): i/o error: End of file
piaware       | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
piaware       | [s6-init] ensuring user provided files have correct perms...exited 0.
piaware       | [fix-attrs.d] applying ownership & permissions fixes...
piaware       | [fix-attrs.d] done.
piaware       | [cont-init.d] executing container ini
piaware       | [cont-init.d] 01-piaware: executing... 
piaware       | Set feeder-id to 16b0b6bc-01d7-454d-8319-2e41976bc832 in /etc/piaware.conf:1
piaware       | warning: could not set option 'allow-auto-updates' to value 'no': cannot update option allow-auto-updates as all config files are read-only
piaware       | warning: could not set option 'allow-manual-updates' to value 'no': cannot update option allow-manual-updates as all config files are read-only
piaware       | warning: could not set option 'allow-mlat' to value 'yes': cannot update option allow-mlat as all config files are read-only
piaware       | warning: could not set option 'mlat-results' to value 'yes': cannot update option mlat-results as all config files are read-only
piaware       | warning: could not set option 'receiver-type' to value 'relay': cannot update option receiver-type as all config files are read-only
piaware       | warning: could not set option 'receiver-host' to value 'readsb': cannot update option receiver-host as all config files are read-only
piaware       | warning: could not set option 'receiver-port' to value '30005': cannot update option receiver-port as all config files are read-only
piaware       | warning: could not set option 'uat-receiver-type' to value 'sdr': cannot update option uat-receiver-type as all config files are read-only
piaware       | warning: could not set option 'uat-receiver-host' to value 'dump978': cannot update option uat-receiver-host as all config files are read-only
piaware       | warning: could not set option 'uat-receiver-port' to value '30978': cannot update option uat-receiver-port as all config files are read-only
piaware       | [cont-init.d] 01-piaware: exited 0.
piaware       | [cont-init.d] done.
piaware       | [services.d] starting services
piaware       | [services.d] done.
piaware       | [beast-splitter] 2022/03/28 19:12:15 127.0.0.1:30004: connection to 127.0.0.1:30004 failed: Connection refused
piaware       | [beast-splitter] 2022/03/28 19:12:15 127.0.0.1:30004: reconnecting in 60 seconds
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[339] W ioctl(5, IOCTL_VM_SOCKETS_GET_LOCAL_CID, ...): Inappropriate ioctl for device
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[339] N listening on AF=2 0.0.0.0:30978
piaware       | [beast-splitter] 2022/03/28 19:12:15 net(readsb:30005): connected to 172.18.0.8:30005
piaware       | [beast-splitter] 2022/03/28 19:12:15 net(readsb:30005): configured with settings: BCDfgIjk
piaware       | [skyaware978] 2022/03/28 19:12:15 Connecting to 127.0.0.1:30978
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[339] N accepting connection from AF=2 127.0.0.1:34574 on AF=2 127.0.0.1:30978
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[339] N forked off child process 354
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[339] N listening on AF=2 0.0.0.0:30978
piaware       | [skyaware978] 2022/03/28 19:12:15 Connected to 127.0.0.1:30978
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[354] N opening connection to AF=2 172.18.0.9:30978
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[354] N successfully connected from local address AF=2 172.18.0.10:47652
piaware       | [978json-splitter] 2022/03/28 19:12:15 2022/03/28 19:12:15 socat[354] N starting data transfer loop with FDs [6,6] and [5,5]
piaware       | [dump1090] 2022/03/28 19:12:16 Mon Mar 28 19:12:16 2022 EDT  dump1090-fa v7.2 starting up.
piaware       | [skyaware] 2022/03/28 19:12:16 2022-03-28 19:12:15: server.c.1513) server started (lighttpd/1.4.59)
[... 9 lines cut ...]
piaware       | [piaware] 2022/03/28 19:12:27 Starting faup1090: /usr/lib/piaware/helpers/faup1090 --net-bo-ipaddr localhost --net-bo-port 30005 --stdout
piaware       | [piaware] 2022/03/28 19:12:27 Started faup1090 (pid 358) to connect to dump1090
piaware       | [piaware] 2022/03/28 19:12:27 UAT support disabled by local configuration setting: uat-receiver-type
piaware       | [piaware] 2022/03/28 19:12:33 Connecting to FlightAware adept server at piaware.flightaware.com/1200
piaware       | [piaware] 2022/03/28 19:12:34 Connection with adept server at piaware.flightaware.com/1200 established
piaware       | [piaware] 2022/03/28 19:12:34 TLS handshake with adept server at piaware.flightaware.com/1200 completed
piaware       | [piaware] 2022/03/28 19:12:34 FlightAware server certificate validated
piaware       | [piaware] 2022/03/28 19:12:34 encrypted session established with FlightAware
piaware       | [piaware] 2022/03/28 19:12:50 Login attempt with adept server at piaware.flightaware.com/1200 timed out
piaware       | [piaware] 2022/03/28 19:12:50 reconnecting in 4 seconds...
piaware       | [piaware] 2022/03/28 19:12:54 Connecting to FlightAware adept server at 206.253.80.199/1200
piaware       | [piaware] 2022/03/28 19:12:54 Connection with adept server at 206.253.80.199/1200 established
piaware       | [piaware] 2022/03/28 19:12:54 TLS handshake with adept server at 206.253.80.199/1200 completed
piaware       | [piaware] 2022/03/28 19:12:54 FlightAware server certificate validated
piaware       | [piaware] 2022/03/28 19:12:54 encrypted session established with FlightAware
piaware       | [piaware] 2022/03/28 19:12:55 logged in to FlightAware as user guest
piaware       | [piaware] 2022/03/28 19:12:55 my feeder ID is b37a45e6-a3ab-46f8-8ef9-a6adad5b0030
piaware       | [piaware] 2022/03/28 19:12:55 site statistics URL: https://flightaware.com/adsb/stats/site/173611
piaware       | [piaware] 2022/03/28 19:12:57 0 msgs recv'd from dump1090; 0 msgs sent to FlightAware
piaware       | [beast-splitter] 2022/03/28 19:13:31 127.0.0.1:30004: connected to 127.0.0.1:30004 with settings 
piaware       | [beast-splitter] 2022/03/28 19:13:31 net(readsb:30005): configured with settings: BCdfGijk
piaware       | [piaware] 2022/03/28 19:13:31 piaware received a message from dump1090!
piaware       | [piaware] 2022/03/28 19:13:32 piaware has successfully sent several msgs to FlightAware!
piaware       | [piaware] 2022/03/28 19:17:57 726 msgs recv'd from dump1090 (726 in last 5m); 726 msgs sent to FlightAware
```

mikenye, 2022-03-29:
There was a bug in the container that was fixed yesterday. Please re-pull and re-create the container. That should solve your problems.

jasonkguinn, 2022-03-29:
Perfect - that definitely resolved the issue. Thanks so much for the fast response and resolution!

closed 2022-03-29 by jasonkguinn

mikenye, 2022-03-29:
No problems, sorry for the bug. Totally my fault. I'm expecting more issues from people in the same position.... 🤦