sdr-enthusiasts/docker-piaware #131: MLAT with docker-piaware using docker-readsb-protobuf not working(?)

SketchCND, 2022-04-25:
I'm using  docker-readsb-protobuf, docker-piaware and other SDR Enthusiasts feeders and am unable to get MLAT working. I've followed the readmes and have also tried using themlathub method but none seem to work. Co-ordinates are set accuratley. I'm in a very populated area with lots of air traffic (near LHR airport) and would expect to synchronization with quite a few MLAT sites.

### My setup:

RPi3 B+ (with official Pi PSU) running Raspberry Pi OS (64bit) and Docker version 20.10.5+dfsg1 and a FlightAware Pro Stick Plus

```
$ uname -a
Linux rpi3-iot 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

$ lscpu
Architecture:                    aarch64
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           4
Model name:                      Cortex-A53
Stepping:                        r0p4
CPU max MHz:                     1400.0000
CPU min MHz:                     600.0000
BogoMIPS:                        38.40
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Not affected
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Not affected
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid

$ lsusb
Bus 001 Device 004: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
Bus 001 Device 005: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ free
               total        used        free      shared  buff/cache   available
Mem:          931588      293208       84288       12768      554092      579532
Swap:         102396       38144       64252

$ docker info
Client:
 Context:    default
 Debug Mode: false

Server:
 Containers: 9
  Running: 9
  Paused: 0
  Stopped: 0
 Images: 34
 Server Version: 20.10.5+dfsg1
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: local
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc io.containerd.runc.v2 io.containerd.runtime.v1.linux
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 1.4.13~ds1-1~deb11u1
 runc version: 1.0.0~rc93+ds1-5
 init version: 
 Security Options:
  seccomp
   Profile: default
  cgroupns
 Kernel Version: 5.15.32-v8+
 Operating System: Raspbian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 909.8MiB
 Name: rpi3-iot
 ID: 6RQR:F6X7:WQ6C:RPQ5:BF57:HRZL:XEF4:GBHW:ICPI:7WDE:ROS3:5SQQ
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No memory limit support
WARNING: No swap limit support
WARNING: Support for cgroup v2 is experimental

$ cat docker-compose.yaml
version: '3.6'

networks:
  adsbnet:

volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:

  readsb:
    image: mikenye/readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - ${DEV_1}
    environment:
      - TZ=${TZ}
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=${LAT}
      - READSB_LON=${LONG}
      - READSB_MAX_RANGE=600
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - READSB_NET_CONNECTOR=mlathub,30105,beast_in
#      - DISABLE_PERFORMANCE_GRAPHS=true
#      - DISABLE_WEBAPP=true

    volumes:
      - ${LOC_TZ}
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb:size=64M
      - /var/log:size=32M
    networks:
      adsbnet:
    ports:
      - 8888:8080
      - 30003:30003
      - 30005:30005
    dns:
      - ${DNS_1}
      - ${DNS_2}

  mlathub:
    image: mikenye/readsb-protobuf:latest
    tty: true
    container_name: mlathub
    restart: always
    environment:
      - TZ=${TZ}
      - DISABLE_PERFORMANCE_GRAPHS=true
      - DISABLE_WEBAPP=true
      - READSB_NET_ENABLE=true
      - READSB_NET_ONLY=true
      - READSB_FORWARD_MLAT=true
      - READSB_NET_CONNECTOR=piaware,30105,beast_in;rbfeeder,30105,beast_in
      - READSB_NET_BEAST_OUTPUT_PORT=30105
      - READSB_NET_BEAST_INPUT_PORT=30104
    volumes:
      - ${LOC_TZ}
    tmpfs:
      - /run/readsb
      - /var/log
    networks:
      adsbnet:
    ports:
      - 30105:30105
    dns:
      - ${DNS_1}
      - ${DNS_2}

  piaware:
    image: mikenye/piaware:latest
    tty: true
    container_name: piaware
    restart: always
    environment:
      - TZ=${TZ}
      - LAT=${LAT}
      - LONG=${LONG}
      - RECEIVER_TYPE=relay
      - BEASTHOST=readsb
      - BEASTPORT=30005
      - ALLOW_MLAT=yes
      - MLAT_RESULTS=yes
      - MLAT_RESULTS_BEASTHOST=mlathub
      - MLAT_RESULTS_BEASTPORT=30104
      - USERNAME=${PiA_USERNAME}
      - PASSWORD=${PiA_PASSWORD}
      - FEEDER_ID=${PiA_FEEDER_ID}
    volumes:
      - ${LOC_TZ}
    tmpfs:
      - /run:exec,size=64M
      - /var/log
    networks:
      adsbnet:
    ports:
      - 8080:80/tcp
    dns:
      - ${DNS_1}
      - ${DNS_2}
    depends_on:
      - readsb
      - mlathub

  fr24feed:
    image: mikenye/fr24feed:latest
    container_name: fr24feed
...

$ docker logs piaware | tail
[piaware] 2022/04/25 09:26:40 69820 msgs recv'd from dump1090 (1299 in last 5m); 69739 msgs sent to FlightAware
[piaware] 2022/04/25 09:31:40 71117 msgs recv'd from dump1090 (1297 in last 5m); 71036 msgs sent to FlightAware
[piaware] 2022/04/25 09:36:40 72363 msgs recv'd from dump1090 (1246 in last 5m); 72282 msgs sent to FlightAware
[piaware] 2022/04/25 09:39:46 mlat-client(2922): Receiver status: connected
[piaware] 2022/04/25 09:39:46 mlat-client(2922): Server status:   clock unstable
[piaware] 2022/04/25 09:39:46 mlat-client(2922): Receiver:  484.2 msg/s received       82.1 msg/s processed (17%)
[piaware] 2022/04/25 09:39:46 mlat-client(2922): Server:      0.0 kB/s from server    0.0kB/s TCP to server     1.0kB/s UDP to server
[piaware] 2022/04/25 09:39:46 mlat-client(2922): Aircraft: 7 of 11 Mode S, 27 of 30 ADS-B used
[piaware] 2022/04/25 09:41:40 73562 msgs recv'd from dump1090 (1199 in last 5m); 73481 msgs sent to FlightAware
[piaware] 2022/04/25 09:46:40 74457 msgs recv'd from dump1090 (895 in last 5m); 74376 msgs sent to FlightAware
```

Message on my FlightWare stats page for my site:
`This feeder is not being used for multilateration because its timing information appears to be unreliable. This can be caused by the site location being incorrect, or because your Pi is running out of free CPU.`

The readsb-protobuf performance graphs show at most 2-3 MLAT messages in total directly after startup but never increment further.

Does anyone have this working in a similar setup? Am I doing something incorrectly that could be causing this.

ebaschiera, 2023-05-08:
Anyone solved this issue in any other way? I am having the same problem, but I am on a fast Intel cpu.
I am using the latest ultrafeeder instead of readsb-protobuf.

Thanks

HanSyt, 2024-12-27:
@ebaschiera check in the docker container with netstat if port 30105 is available. (docker exec piaware netstat)
Then check utltrafeeder if something goes wrong with the connection of piaware. (docker logs ultrafeeder | grep piaware).

I noticed 2 things:
 - port 30105 on piaware was refusing connections, redeploy of the stack solved this
 - port 30105 on piaware simply did not excist: my stupid mistake, wrong feeder ID (did not copy the first character)

ebaschiera, 2024-12-27:
Thank you for your reply, but I gave up and switched to https://github.com/dirkhh/adsb-feeder-image on a small dedicated PC.
No more issues!