sdr-enthusiasts/docker-piaware #133: support for SSL-/-TLS certificates

bartgrefte, 2022-08-05:
Just out of curiosity, why doesn't there seem to be support for SSL-/TLS-certificates?

I have yet to find even óne docker container for ADS-B where the webinterface to display the flights can be used with those.

Consider this a request to implement support for usage of certificates.

needs-coffee, 2022-08-31:
@bartgrefte - The typical and appropriate method to achieve TLS for docker containers is to use a webserver as reverse proxy such as nginx/traefik/caddy and proxy the connection to the container. You can do this within the same docker network and only expose the webserver container. 

As these are established and tested webservers they are usually better performing and more secure at implementing TLS.

bartgrefte, 2022-09-14:
@needs-coffee I am aware of that, unfortunately, I have yet to find a pre-made reverse proxy container (like Nginx Proxy Manager) that supports adding pre-existing certificates through command line.

Right now I've set up an Apache webserver and manually configured that to function as a reverse proxy, where I can use my own pre-existing certificates. Not quite user-friendly, fortunately I have worked with Apache config-files before, so it is doable.

needs-coffee, 2022-09-14:
I use nginx but i dont use nginx proxy manager - i just edit the config files directly, more portable. 
If you use docker compose you can add this 
```YAML
services:
  reverse_proxy:
    container_name: reverse_proxy
    image: nginx
    ports:
      - 80:80
      - 443:443
    volumes:
      - ./config/default.conf:/etc/nginx/conf.d/default.conf
      - ./certs:/etc/nginx/certs
    networks:
      - adsb
    depends_on:
      - readsb
  ```

and in the docker-compose directory - `./config/default.conf`
```CONF
server {
    listen       80 default_server;
    listen  [::]:80 default_server;
    return 302 https://$host$request_uri;
}

upstream readsbcontainer {
    server readsb:8080;
}

server {
    listen 443              ssl http2 default_server;
    listen [::]:443         ssl http2 default_server;
    # omit server_name for catch all default server, if redirecting multiple subdomains specify server name
    # server_name           example.lan example.local;

    ssl_certificate         /etc/nginx/certs/mycertchain.pem;
    ssl_certificate_key     /etc/nginx/certs/mycert.key;
    ssl_trusted_certificate /etc/nginx/certs/myrootcert.pem;

    ssl_protocols           TLSv1.2 TLSv1.3;

    error_page   500 502 503 504  /50x.html;
    location = /50x.html {
        root   /usr/share/nginx/html;
    }

    location / {
        proxy_set_header        Host $host;
        add_header              X-Served-By $host;
        proxy_set_header        X-Real-IP $remote_addr;
        proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header        X-Forwarded-Proto $scheme;
        proxy_set_header        X-Forwarded-Scheme $scheme;
        proxy_read_timeout      90;
        proxy_http_version      1.1;
        proxy_pass              http://readsbcontainer/;
        proxy_redirect          http://readsbcontainer https://$host;
        }

}```

Certs are added in a certs in the docker-compose folder  with your chain, key and root ca 
I use nginx on the host not in a container though so i can proxy non docker connections 
I would imagine traefik would be easier as it is 'docker native'

mikenye, 2023-02-09:
Please see: https://github.com/sdr-enthusiasts/docker-reversewebproxy

closed 2023-02-09 by mikenye