sdr-enthusiasts/docker-piaware #142: Site Location / Range rings missing in 8.2

speedst3r, 2023-04-12:
I am using the piaware v8.2 amd64 container and have noticed my site location and range rings are not displaying. I have unset the LAT/LONG environment variables to no effect. My understanding is with 8.2 it is meant to get the site information from piaware itself, and it looks like it is.

https://pastebin.com/DcwjiCr6

If I set the LAT/LONG env vars and downgrade to 8.1 this works successfully. Is there some additional work required for this container to get location/range rings on 8.2?

speedst3r, 2023-04-15:
Rings visible in 8.1
[image]

Rings missing in 8.2
[image]

speedst3r, 2023-04-15:
Tested with truecharts/piaware:v8.2.0 container (that still accepts LAT/LONG env vars) and station/rings are visible.

speedst3r, 2023-04-16:
I built this container locally after reverting f7e57dd (and supplying LAT/LONG environment vars); the station location and rings returned.

seichold, 2023-07-03:
Also impacts me. Also breaks UAT range estimations when using ultrafeeder (stats graph for UAT range is just blank). While this works for feeding flight aware (mlat etc works), it breaks other things locally. Please allow these again at least as a optional env.

Tagliax, 2024-01-17:
Same problem here, on version 9.0
I fixed this problem editing config.js in /usr/share/skyaware/html.
More precisely I set the parameter SiteShow to "true".
For the localization to be correct, also change the location coordinates both here and in the config.js in the dump1090/html folder.
Restart container and clear web cache.
[image]

speedst3r, 2024-08-17:
This seems to have been ignored... reported 16 months ago and no movement.

@mikenye any chance this could get looked at?

closed 2024-10-20 by fredclausen

wiedehopf, 2024-10-20:
This change will use the location as configured on the FA stats page.
Does that work for everyone?

speedst3r, 2024-10-21:
> Does that work for everyone?

Not working for me currently on container start, however does work if I use `s6-svc` to restart the `dump1090` service.

On container launch the dump1090 command is

`/usr/local/bin/dump1090 --quiet --net --fix --json-location-accuracy 2 --write-json /run/dump1090-fa --net-bind-address 0.0.0.0 --net-only`

After restarting the service it is

`/usr/local/bin/dump1090 --quiet --net --fix --json-location-accuracy 2 --write-json /run/dump1090-fa --net-bind-address 0.0.0.0 --lat <nope> --lon <nope> --net-only`

Seems like `location.env` is not there fast enough when the container starts. Is there some sort of race condition?

wiedehopf, 2024-10-21:
It's not a race.
Once piaware gets the location from the FA server, piaware restarts dump1090.

speedst3r, 2024-10-21:
> If i understand you correctly you're just not waiting long enough.

This on top of stating nobody raised a PR to fix - is blaming users the standard here?

It would seem piaware is unable to restart dump1090 for some reason (08:38:18.215).

```
s6-rc: info: service dump1090: starting
s6-rc: info: service dump1090 successfully started
/etc/s6-overlay/scripts/dump1090: line 37: /var/cache/piaware/location.env: No such file or directory
[2024-10-21 08:37:31.421][dump1090] Mon Oct 21 08:37:31 2024 AWST  dump1090-fa v9.0 starting up.
[2024-10-21 08:37:31.423][dump1090] Net-only mode, no SDR device or file open.
[2024-10-21 08:37:34.351][piaware] ADS-B data program 'dump1090' is listening on port 30005, so far so good
[2024-10-21 08:37:34.354][piaware] Started faup1090 (pid 223) to connect to dump1090
[2024-10-21 08:37:36.060][piaware] piaware received a message from dump1090!
[2024-10-21 08:38:05.827][piaware] 63 msgs recv'd from dump1090; 0 msgs sent to FlightAware
[2024-10-21 08:38:18.036][piaware] Receiver location changed, restarting dump1090 and skyaware978
[2024-10-21 08:38:18.036][piaware] attempting to restart dump1090..
[2024-10-21 08:38:18.215][piaware] can't restart dump1090, no services that look like dump1090 found
[2024-10-21 08:38:19.226][piaware] reconnecting to dump1090
[2024-10-21 08:38:20.415][piaware] ADS-B data program 'dump1090' is listening on port 30005, so far so good
[2024-10-21 08:38:20.415][piaware] Started faup1090 (pid 249) to connect to dump1090
```

wiedehopf, 2024-10-21:
For me it's restarting dump1090 just fine.
```
[2024-10-21 00:42:11.582][piaware] Receiver location changed, restarting dump1090 and skyaware978
[2024-10-21 00:42:11.583][piaware] attempting to restart dump1090.. 
[2024-10-21 00:42:11.584][piaware] has_invoke_rcd: Running on docker ignoring invoke-rc.d
[2024-10-21 00:42:11.585][piaware] has_invoke_rcd: Running on docker ignoring invoke-rc.d
[2024-10-21 00:42:11.585][piaware] attempting to restart dump1090 using '/etc/init.d/dump1090 restart < /dev/null &'...
[2024-10-21 00:42:11.588][piaware] dump1090 restart appears to have been successful
[2024-10-21 00:42:11.589][piaware] attempting to restart skyaware978.. 
[2024-10-21 00:42:11.590][piaware] has_invoke_rcd: Running on docker ignoring invoke-rc.d
[2024-10-21 00:42:11.590][piaware] has_invoke_rcd: Running on docker ignoring invoke-rc.d
[2024-10-21 00:42:11.590][piaware] attempting to restart skyaware978 using '/etc/init.d/skyaware978 restart < /dev/null &'...
[2024-10-21 00:42:11.593][piaware] skyaware978 restart appears to have been successful
```

Let's check this:
```
docker exec -it piaware ls /etc/init.d -l
```

wiedehopf, 2024-10-21:
Hmmm i think it's cause one file in specific might not exist for you?
```
docker exec -it piaware ls -l /.dockerenv
```

wiedehopf, 2024-10-21:
Hmm some googling seems to say /.dockerenv is not really a thing.
Might as well hard-code it as that file is changed for docker anyhow.

wiedehopf, 2024-10-21:
Changed version that doesn't rely on /.dockerenv is building, will be done in 30 min.
Let's see if that fixes it.

speedst3r, 2024-10-21:
https://github.com/sdr-enthusiasts/docker-piaware/pkgs/container/docker-piaware/292224492?tag=latest-build-555 restarts dump1090 successfully - thanks for the quick triage and turnaround!