sdr-enthusiasts/docker-piaware #143: docker-piaware container issues in ADSB.IM

chevyman142000, 2023-07-12:
Hello - I am hoping to get some help with this container. I'm using the ADSB.IM image to feed a bunch of different aggregators as well as FlightAware. I've documented the issue with Dirk/ADSB.IM on GitHub here, so I'm hoping this has enough detail of the issue I'm having. https://github.com/dirkhh/adsb-feeder-image/issues/20 Some additional details, I'm seeing it disconnect from Flightaware probably every 5 minutes. I originally had this setup on a Pi 3B+ with an old SD card. I have since set it up from scratch with a brand new Sandisk sd card, Pi 4 and brand new Pi USB-C power adapter. Please let me know what other information I can get to help troubleshoot this. I appreciate any help or thoughts on this!

TLDR: PiAware container loses connection to FlightAware every ~5 minutes and eventually consumes 100% of 1 core and stops uploading to FlightAware.

wiedehopf, 2024-10-20:
This still an issue or can it be closed?

chevyman142000, 2024-10-21:
I don't believe this is an issue anymore. Thanks!

closed 2024-10-21 by chevyman142000