sdr-enthusiasts/docker-piaware #150: AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'

idatum, 2023-10-21:
I periodically see an exception running the latest image with piaware 8.2. Looks like a module version mismatch? I could not find any issue on piaware's repo. Here are my Docker logs:

[piaware] 2023/10/21 19:06:07 mlat-client(174): outlier detected with ts: 6269.091, last_ts: 6269.063, ts_elapsed: 0.027, sys_elapsed: 1.503 (values in seconds)
[piaware] 2023/10/21 19:06:07 mlat-client(174): Unexpected exception on connection to localhost:30005
[piaware] 2023/10/21 19:06:07 mlat-client(174): Traceback (most recent call last):
[piaware] 2023/10/21 19:06:07 mlat-client(174): File "/usr/lib/python3.11/asyncore.py", line 88, in read
[piaware] 2023/10/21 19:06:07 mlat-client(174): obj.handle_read_event()
[piaware] 2023/10/21 19:06:07 mlat-client(174): File "/usr/lib/python3.11/asyncore.py", line 425, in handle_read_event
[piaware] 2023/10/21 19:06:07 mlat-client(174): self.handle_read()
[piaware] 2023/10/21 19:06:07 mlat-client(174): File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/receiver.py", line 214, in handle_read
[piaware] 2023/10/21 19:06:07 mlat-client(174): self.coordinator.input_received_messages(messages)
[piaware] 2023/10/21 19:06:07 mlat-client(174): File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 328, in input_received_messages
[piaware] 2023/10/21 19:06:07 mlat-client(174): handler(message, now)
[piaware] 2023/10/21 19:06:07 mlat-client(174): File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 350, in received_timestamp_jump_event
[piaware] 2023/10/21 19:06:07 mlat-client(174): self.server.send_clock_jump()
[piaware] 2023/10/21 19:06:07 mlat-client(174): ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[piaware] 2023/10/21 19:06:07 mlat-client(174): AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'
[piaware] 2023/10/21 19:06:07 mlat-client(174): Lost connection to localhost:30005

mutability, 2023-10-22:
this is caused by https://github.com/sdr-enthusiasts/docker-piaware/commit/e305643e670cc41da99d32a13ab49b097648e6a7 -- the Dockerfile is using the wrong mlat-client. piaware is not expecting / does not support use of wiedehopf's fork, please use mutability/mlat-client following what piaware_builder does.

mutability, 2023-10-22:
I also opened https://github.com/wiedehopf/mlat-client/pull/2 to see if we can reduce the confusion with different forks of mlat-client a bit.

kx1t, 2023-11-11:
fixed in https://github.com/sdr-enthusiasts/docker-piaware/commit/11571193a55cd6cb65a6b85e6e272cdcd628a5c3

Thanks for doing the sleuthing to find which commit introduced the issue. 

> this is caused by [e305643](https://github.com/sdr-enthusiasts/docker-piaware/commit/e305643e670cc41da99d32a13ab49b097648e6a7) -- the Dockerfile is using the wrong mlat-client. piaware is not expecting / does not support use of wiedehopf's fork, please use mutability/mlat-client following what piaware_builder does.

closed 2023-11-11 by kx1t

fredclausen, 2023-11-11:
Thanks for finding the issue so we could revert. Your version of the MLAT client was broken in bookworm based debian's - like our containers because the python version was "too new" until [this commit](https://github.com/mutability/mlat-client/commit/bf048ea4c679dc6ff9abaf7e8d2405e4a7aa8aab) was merged in. Switching client's fixed the issue.

Sorry if this caused any issues for FlightAware MLAT results, but I wasn't aware wiedehopf's fork was "wrong" and that was patched appropriately to work on the version of python included in bookworm releases.

idatum, 2023-11-19:
This is still an issue running v9.0

[piaware] 2023/11/19 21:28:42 mlat-client(174): outlier detected with ts: 34.368, last_ts: 99.340, ts_elapsed: -64.972, sys_elapsed: 2242.911 (values in seconds)
[piaware] 2023/11/19 21:28:42 mlat-client(174): Unexpected exception on connection to localhost:30005
[piaware] 2023/11/19 21:28:42 mlat-client(174): Traceback (most recent call last):
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 88, in read
[piaware] 2023/11/19 21:28:42 mlat-client(174):     obj.handle_read_event()
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 425, in handle_read_event
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.handle_read()
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/receiver.py", line 214, in handle_read
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.coordinator.input_received_messages(messages)
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 328, in input_received_messages
[piaware] 2023/11/19 21:28:42 mlat-client(174):     handler(message, now)
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 350, in received_timestamp_jump_event
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.server.send_clock_jump()
[piaware] 2023/11/19 21:28:42 mlat-client(174):     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[piaware] 2023/11/19 21:28:42 mlat-client(174): AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'
[piaware] 2023/11/19 21:28:42 mlat-client(174): Lost connection to localhost:30005

fredclausen, 2023-11-19:
Just to be sure you've updated to the latest container?

idatum, 2023-11-19:
Yes, my docker-compose.yaml:
image: ghcr.io/sdr-enthusiasts/docker-piaware:v9.0

docker logs piaware:
[piaware] 2023/11/19 22:10:45 piaware version 9.0 is running, process ID 143