sdr-enthusiasts/docker-piaware #156: piaware v9.0: AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'

idatum, 2023-11-19:
This is using ghcr.io/sdr-enthusiasts/docker-piaware v9.0:

[piaware] 2023/11/19 21:28:42 mlat-client(174): outlier detected with ts: 34.368, last_ts: 99.340, ts_elapsed: -64.972, sys_elapsed: 2242.911 (values in seconds)
[piaware] 2023/11/19 21:28:42 mlat-client(174): Unexpected exception on connection to localhost:30005
[piaware] 2023/11/19 21:28:42 mlat-client(174): Traceback (most recent call last):
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 88, in read
[piaware] 2023/11/19 21:28:42 mlat-client(174):     obj.handle_read_event()
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/lib/python3.11/asyncore.py", line 425, in handle_read_event
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.handle_read()
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/receiver.py", line 214, in handle_read
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.coordinator.input_received_messages(messages)
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 328, in input_received_messages
[piaware] 2023/11/19 21:28:42 mlat-client(174):     handler(message, now)
[piaware] 2023/11/19 21:28:42 mlat-client(174):   File "/usr/local/lib/python3.11/dist-packages/MlatClient-0.4.2-py3.11-linux-x86_64.egg/mlat/client/coordinator.py", line 350, in received_timestamp_jump_event
[piaware] 2023/11/19 21:28:42 mlat-client(174):     self.server.send_clock_jump()
[piaware] 2023/11/19 21:28:42 mlat-client(174):     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[piaware] 2023/11/19 21:28:42 mlat-client(174): AttributeError: 'AdeptConnection' object has no attribute 'send_clock_jump'
[piaware] 2023/11/19 21:28:42 mlat-client(174): Lost connection to localhost:30005

fredclausen, 2023-11-19:
the `v9.0` tag hasn't been updated for about 13 days, which if memory serves was before we fixed the issue. Looks like piaware updated their software to 9.0.1 and we automatically generated a new tag for it. I would suggest `latest` as the tag you use.

idatum, 2023-11-19:
Thanks, confirmed using :latest tag got me to 9.0.1:
2023/11/19 23:00:50 piaware version 9.0.1 is running, process ID 140
...
[piaware] 2023/11/19 23:11:25 1518 msgs recv'd from dump1090 (935 in last 5m); 1518 msgs sent to FlightAware
[piaware] 2023/11/19 23:15:53 mlat-client(174): Receiver status: connected

closed 2023-11-19 by idatum

fredclausen, 2023-11-19:
Awesome! I'm glad it is working, and thanks again for bringing the issue up so we could fix it.