sdr-enthusiasts/docker-piaware #221: PIAware in Beast mode not receiving data

HanSyt, 2024-12-26:
I have docker setup with a rtlsdr receiver
https://github.com/sdr-enthusiasts/docker-readsb-protobuf
This works, I have a map with flight data of aircraft

I added  adsb exchange support to it:
https://github.com/sdr-enthusiasts/docker-adsbexchange
This works, system starts and turns healthy when the first packets come in.

However, I did setup the flight aware docket container
https://github.com/sdr-enthusiasts/docker-piaware
This one starts and keeps stating it is starting, looking in the container I see that all seems to be allright, except no data is sent to piaware. Looking at the map, no data is on the map as well (-p 80:8080).

[image]

This is my docket up config for PI-Aware
[image]

What am I doing wrong,
[image]

wiedehopf, 2024-12-26:
Up to date guide: https://sdr-enthusiasts.gitbook.io/ads-b/

As for piaware read this: https://github.com/sdr-enthusiasts/docker-piaware?tab=readme-ov-file#determining-your-feeder-id

Or is the issue just that you put the id in acute braces? <> Don't do that.

HanSyt, 2024-12-26:
@wiedehopf , Thanks, Feeder ID is ok (nu braces used). However, I had the idea it was not connecting to beasthost (without uploading tot piaware, you should see the planes on the map).

So I changed BEASTHOST= from beasthost to readsb, no change, then the IP of it: 127.17.0.4 and it jolly went up and running. Little bit strange however, adbsx had no problems with beashost..

[image]

[image]

wiedehopf, 2024-12-26:
Look at the guide with ultrafeeder and feeding a couple more sites like airplanes.live / adsb.fi / adsb.lol :)
adsbx can also be fed from ultrafeeder.

readsb protobuf container isn't really maintained, better to use ultrafeeder in its place.

BEASTHOST always needs to point to the data source container, it can't work pointing at beasthost.
just because the container is showing healthy wouldn't necessarily mean it's getting data.

If this solves your issue feel free to close.

HanSyt, 2024-12-26:
Maybe ADBSX is not sending data, I have to check that, meanwhile I changed all beasthosts to the host IP.
Both are heathy now. Thanks so far

closed 2024-12-26 by HanSyt

wiedehopf, 2024-12-26:
You should be able to use the container name as a hostname.
If you use docker run, be sure to use --hostname to set that.

Using the IP is not a stable setup.

wiedehopf, 2024-12-26:
Also to repeat, those are somewhat unmaintained containers (readsb protobuf / adsbx)

Just using ultrafeeder will mean you don't need the extra adsbx container as feeding that site is included in ultrafeeder just like feeding a bunch of other sites.

HanSyt, 2024-12-26:
Concerning the hostname, it is readsb. If I go into the readsb container en exec a ping readsb I get a nice response back.
However If I go to the other 2 containers and do a ping readsb, no response. So I use the host IP now (and port forwarding to the host for mlat). So far about this type of feeding.

I just had a quick look at Ultrafeeder, looks nice. However the other system made it possible to check one feeders separate from each other (somehow easier). But I wil try utlrafeeder (.env en compose.yml are downloaded already.

Already have a raspberry py (3b) up and running with piaware and adsexchange uploads. This one is supposed to replace the old one that's running for 3 years now.

wiedehopf, 2024-12-26:
Maybe you specified a specific network?
The containers need to be on the same docker network.

The FA container will stay separate anyhow.
And really if the ultrafeeder is configured correctly and the feed doesn't work, that's usually on the site.

HanSyt, 2024-12-26:
@wiedehopf Got utlrafeed working. Configuring is not that easy, I guess 
ADSB_SDR_SERIAL=1
ADSB_SDR_PPM=
maybe need some extra explanation, it is all in the readme.md, but a bit fragmented.

After that I needed this FA containers as well to feed FLightRadar as you mentioned (utlrafeed needed an extra -p 30005:30005 and -p 30105:30105 to the host to get them together. 

The picture below is probably what you ment with: "that's usually on the site."

[image]

HanSyt, 2024-12-27:
And the other riddle is solved too. If alle feeders/beasts are on the same network they should be able to ping/find each other by their name. Except when they are all on the same default bridge! SInce utlrafeeder made its own network I had to add --net to the piaware deployment script

[image]

wiedehopf, 2024-12-27:
Really everyone uses the compose examples, not sure why you're using docker run.

Feel free to make some PRs to clean up the examples preferrably to just use the default network and not some specific network.

HanSyt, 2024-12-27:
I was used to docker run, but I made one single compose file now with ultrafeeder and piaware in it. Even defined the network, so it is a nice stack now. 
arp can now find the partners in the docker network (adsbnet)

[image]