sdr-enthusiasts/docker-piaware #233: MLAT UDP-Port changing at every Container-Restart

DVTrast, 2025-02-16:
I´m running Piaware in an docker-container. I´ve enabled MLAT but everytime the container get´s restarted, the destination UDP-Port and IP-Address of the MLAT changes. This is not working in my environment since all needed outgoing UDP-Port´s have to be enabled on the firewall. Is there an way to configure a specific UDP-Port for that? I´ve tried "MLAT_RESULTS_BEASTPORT" but this setting worked for me only together with the setting "MLAT_RESULTS_BEASTHOST" - but here I´m unable to find a list of the needed IP-Addresses.
Looking forward to hearing from you.
Best regards

wiedehopf, 2025-02-16:
<https://www.flightaware.com/adsb/faq/>

```
If you have a newer FlightFeeder (supplied from year 2016 forward), the FlightFeeder needs to be able to make the following outbound connections from your local network to the Internet:

UDP port 1194 or TCP port 443 outgoing to ovpn-1.hou.flightaware.com
TCP port 443 outgoing to flightaware.com
TCP port 1200 outgoing
UDP to a port in the range 4999-19999 outgoing
```
Pretty sure this applies here.

See also: <https://discussions.flightaware.com/t/flightaware-mlat-udp-using-wrong-port-range/65353>

This can't be configured.

DVTrast, 2025-02-17:
@wiedehopf Thank you for you information and links.

closed 2025-02-17 by DVTrast